zvldz/mgl03_fw #2: monit binary

kkmaslowski, 2021-04-07:
Hi. Please add monit binary to your firmware https://mmonit.com/monit/

zvldz, 2021-04-07:
You can try 
```
wget "http://master.dl.sourceforge.net/project/mgl03/bin/monit-static?viasf=1" -O /data/monit && chmod +x /data/monit
```
Then I will probably add it to the firmware.

kkmaslowski, 2021-04-07:
Hi. Great, got it working.

Temporary config file (/data/monitrc):
```
set daemon 60
set logfile syslog facility log_daemon
set eventqueue
        basedir /data/.monit/eventqueue
        slots 100
set statefile /data/.monit/state
set idfile /data/.monit/id
set pidfile /data/.monit/pid
set httpd port 2812
        allow localhost
```

Create /data/.monit dir and start service: 
`/data/monit -c /data/monitrc
`

closed 2021-04-08 by kkmaslowski