zvldz/mgl03_fw #4: how to rollback to factory default firmware?

kingwap99, 2021-05-06:
I just installed the firmware provided here.
but there seems something wrong, because I cannot add any zigbee devices from HA and Mi home app.
mi home mode, zha mode and zigbee2mqtt, the three modes not working for me. 
so Is there any way to rollback to factory default firmware ?
maybe I can make another try from there, thanks.

zvldz, 2021-05-06:
You can install custom firmware mgl03_1.4.6_0012. And then update via MiHome

kingwap99, 2021-05-06:
> You can install custom firmware mgl03_1.4.6_0012. And then update via MiHome

if update from MiHome app,  it failed and left a lot of big files in mi gateway 2

zvldz, 2021-05-06:
delete unnecessary large files, if you are using a component for HA, turn off  update lock.

zvldz, 2021-05-06:
All * .bin files can be deleted, except for hw.bin

kingwap99, 2021-05-06:
> delete unnecessary large files, if you are using a component for HA, turn off update lock.

thanks for your quick reply, for sure already clean unecessory files, and turn off update lock

======================================================
FIRMWARE........: 1.4.6_0012 #20210309
IP..............: 192.168.31.246
MAC.............: 54:EF:44:CF:95:11
TOKEN...........: 5343577676504b76677979613371336b
DID.............: 374719075
KEY.............: EyW6BQItjv5YwD6A
======================================================

Filesystem           1024-blocks    Used Available Use% Mounted on
/dev/root                11136     11136         0 100% /
ubi0_0                   46904      2552     41920   6% /data
/data/miio               46904      2552     41920   6% /data/miio

zvldz, 2021-05-06:
What is the error when updating via MiHome?
Have you updated the zigbee fw?

kingwap99, 2021-05-06:
> What is the error when updating via MiHome?
"Upgrade failed, please retry later"

> Have you updated the zigbee fw?
Do you mean zegbee device's fw, no I didn't.
actually,  I have another one mi gateway 2 work under zha mode very smoothly, 
so I can make sure all the zigbee devices can work well.

zvldz, 2021-05-06:
It is very likely that the update lock remains. Make sure there are no files:
```
/data/firmware.bin
/data/firmware/firmware_ota.bin
```
If the files exist, you should try to delete them.
If the files are not deleted, then the blocking is enabled.

kingwap99, 2021-05-06:
> /data/firmware/firmware_ota.bin

yes, I found this file and already removed it, but still got the same error message 
"Upgrade failed, please retry later"

the file list below is what I made before this time upgrade.

# cd data
# ls -Rl
.:
drwxrwxrwx    2 app      0              160 Apr 30 07:20 basic_app
drwxrwxr-x    2 admin    0              240 Jan  1  1970 basic_gw
-rwxr-x---    1 admin    0               73 Jan  1  1970 ble_info
-rwxr-xr-x    1 admin    0          1576152 Apr 29 03:06 busybox
drwxr-xr-x    2 admin    0              520 Jan  1  1970 conf
drwxr-xr-x    2 admin    0              320 Jan  1  1970 factory
drwxr-xr-x    2 admin    0              160 May  7 03:37 firmware
drwxr-xr-x    2 admin    0              224 Jan  1  1970 hw_setting
drwxr-xr-x    2 admin    0              888 May  7 03:44 miio
-rw-r--r--    1 admin    0            12288 Jan  1  1970 miioconfig.db
drwx------    2 admin    0              232 Mar 31 00:04 mijia_automation
-rw-r--r--    1 admin    0                0 Jan  1  1970 mount_ok
-rw-r-----    1 admin    0               90 Jan  1  1970 passwd
-rw-r-----    1 admin    0               90 Jan  1  1970 passwd-
-rw-r-----    1 admin    0               68 Jan  1  1970 shadow
-rw-r-----    1 admin    0               68 Jan  1  1970 shadow-
-rwxr-xr-x    1 admin    0           946604 May  7 02:31 silabs_ncp_bt
drwxr-xr-x    3 admin    0              448 Feb 16 02:10 silicon_zigbee_host
-rwxr-xr-x    1 admin    0           422976 Apr 28 19:49 socat
drwxr-xr-x    2 admin    0              624 Apr 28 22:14 zigbee
drwxr-xr-x    2 admin    0              312 Jan  1  1970 zigbee_gw
drwxr-xr-x    2 admin    0              232 May  4 07:00 zoneinfo

./basic_app:

./basic_gw:
-rw-rw-rw-    1 admin    0               12 Jan  1  1970 bind_succeed.flag

./conf:
-rw-rw-r--    1 admin    0              998 May  4 06:59 hostapd3.conf
----------    1 app      20063846        74 Jan  1  1970 lumi.conf
----------    1 app      20080230        64 Jan  1  1970 os-release
-rw-r--r--    1 admin    0                0 Jan  1  1970 set_firmware_ok
-rw-r--r--    1 admin    0              154 Jan  1  1970 wpa-wpa2.conf

./factory:
-rw-r--r--    1 admin    0                9 Jan  1  1970 factory_wifi_passwd
-rw-r--r--    1 admin    0                9 Jan  1  1970 factory_wifi_ssid

./firmware:

./hw_setting:
---------x    1 admin    0          5290602 Jan  1  1970 hw.bin

./miio:
-rw-r--r--    1 admin    0            12288 Apr 29 22:50 HistoryDevice.db
-rw-r--r--    1 admin    0            12288 May  4 07:01 bt_kv_db
-rw-r--r--    1 admin    0               94 Jan  1  1970 device.conf
-rw-r--r--    1 admin    0               17 May  4 06:59 device.token
-rw-r--r--    1 admin    0               10 May  4 07:00 device.uid
-rw-r--r--    1 admin    0              850 Jan  1  1970 lumi_hkinfo.bin
-rw-r--r--    1 admin    0            69632 May  7 03:44 mible_local.db
-rw-r--r--    1 admin    0                2 May  7 02:26 ota_state.txt
-rw-r--r--    1 admin    0              850 Jan  1  1970 setup_bak.conf
-rw-r--r--    1 admin    0               63 May  4 07:00 wifi.conf

./mijia_automation:
-rw-r--r--    1 admin    0            12288 Mar 31 00:04 db.unqlite

./silicon_zigbee_host:
-rw-r--r--    1 admin    0              145 Apr 28 22:14 devices.txt
-rw-r--r--    1 admin    0            13413 Jan  1  1970 gateway.log
-rw-r--r--    1 admin    0               41 Mar 18 01:42 gatewayInfo.txt
drwx------    2 admin    0              240 May  7 03:41 ota-files

./silicon_zigbee_host/ota-files:
-rw-r--r--    1 admin    0           184786 May  7 03:41 ota-file-0001-655.ota

./zigbee:
-rw-r--r--    1 admin    0              147 Jan  1  1970 coordinator.info
-rw-r--r--    1 admin    0               14 Apr 28 22:14 device.info
-rw-r--r--    1 admin    0               23 Apr 28 22:14 deviceObject.info
-rw-r--r--    1 admin    0               18 Apr 28 22:14 fastJoindev.info
-rwxr-x---    1 admin    0              698 Jan  1  1970 miio.info
-rw-r--r--    1 admin    0              727 Apr 28 22:14 zigbeeAgent.info

./zigbee_gw:
-rw-r--r--    1 admin    0              115 Apr 28 22:14 device_properties.json
-rw-r--r--    1 admin    0             8192 Jan  1  1970 zigbee_gw.db

./zoneinfo:
lrwxrwxrwx    1 admin    0               31 May  4 07:00 localtime -> /usr/share/zoneinfo/Asia/Taipei

zvldz, 2021-05-07:
I don't know why the problem occurs. 
You can try manually flashing the stock fw. I'll upload stock fw later.

zvldz, 2021-05-07:
stock fw
https://github.com/zvldz/mgl03_fw/tree/main/firmware/stock/mgl03_1.4.6_0012

kingwap99, 2021-05-07:
thanks for your support, 
but I just fund one thing.
I purchased two mi gateway 2 from different resources 
the first one with model name: ZNDMWG03LM can work very well, but 
the second with model name: ZNDMWG02LM which we are discussing on now.
Does this matter?

zvldz, 2021-05-07:
version should not matter
maybe broken nand
try updating manually
and try installing stock zigbee fw
https://github.com/zvldz/mgl03_fw/tree/main/zigbee

zvldz, 2021-05-07:
and try to look
```
/var/log/miio_client.log
/var/log/messages
```

zvldz, 2021-07-14:
https://github.com/zvldz/mgl03_fw/commit/0c39d87b3884634e25a114af94f5259645888ab7

closed 2021-07-14 by zvldz