zvldz/mgl03_fw #8: I don't know what have I done,

hamwong, 2021-08-24:
I have upgrade to modified FW mgl03_1.5.0_0026,
I tried find some sh by **find / -name .sh** and run it then I found my gateway was offline from MiApp, 
my gateway LED no longer in blue, it's in yellow blinking, how can I fallback?

also why telnet will get freeze if I run some command, or even only **ls /bin** or cat a file?

thank you

zvldz, 2021-08-24:
This is not normal.
These same commands work fine for me.
Maybe you have a problem with nand.

hamwong, 2021-08-24:
but do you have idea why my gw keep offline on Miapp (I still able to mqtt and telnet to GW) and LED keep blue?
I am sure it should cause by one of the shell script that I ran

zvldz, 2021-08-24:
If you restart gateway, it appears in MiHome ?

hamwong, 2021-08-24:
NO I tried soft and hard reset all didn't work,
I tried power off for around 1 min, still won't work
and no matter what, my led just won't turn blue,

I also found if I connect by ssh, I won't get freeze in most of the case, but still happens on telnet,

I also found my system time doesn't right
# date
Thu Jan  1 09:34:59 HKT 1970

zvldz, 2021-08-24:
time should be updated via Internet
Maybe  gateway has problem with  Internet.
Then it will not be available in MiHome.
You can also join to our telegram group https://t.me/xiaomi_gw_v3_hack

Maybe someone else can help you

hamwong, 2021-08-24:
thank you, I will join

closed 2021-09-10 by zvldz