zvldz/mgl03_fw #10: A few questions

ldwnt, 2021-09-05:
1. What happens if I update the firmware of the gateway in Mi home? Will the hack be gone and I have to wait for the new version of hack? Currently the connections are stable with version 1.5.0_0102, just curious about how the firmware update works.

2. Currently the latest version is 150, but according to the scripts below, version 123 is used instead. Is it intended?
```
BLE_VER=$(grep -oe '1\.[23]\..' /tmp/full*gbl | sed 's/\.//g')
if [ -z $BLE_VER ]; then
    echo "! BLE firmware version is not detected. Use 125."
    BLE_VER=123
fi
```

3. I take a look at the shell and it seems what matters is the invocation of run_ble_dfu.sh and fw_update. So maybe on the hard way I just need to wget the 3 files from an outer web server and flush them?

zvldz, 2021-09-09:
If you install the firmware through MiHome, you will have the original firmware.
The scripts here calculate the version of the bluetooth firmware, it has no relationship with the general version

ldwnt, 2021-09-10:
Thanks for explanation