zvldz/mgl03_fw #11: No space under /tmp to use

ldwnt, 2021-09-09:
First I wget the firmware zip to /tmp and it stucks during unzip. Command df shows there is no space anymore. Strange thing is that I can not find the zip file or the unzipped (if any) files under /tmp. Is it safe to leave mount point / with no space?

Can I use /data/miio to unzip the firmware zip file and flush the bin files from there?

```
Filesystem           1024-blocks    Used Available Use% Mounted on
/dev/root                10368     10368         0 100% /
ubi0_0                   46904     12628     31848  28% /data
/data/miio               46904     12628     31848  28% /data/miio
```

zvldz, 2021-09-09:
The root partition has no free space because it is squashfs and read-only.
The /tmp partition is insufficient for zip itself and for unpacking it.
You can download the archive to /data, and unpack to /tmp or to /data .
But it's better to use a script that will do everything yourself

zvldz, 2021-09-09:
If there is not enough space in /data, then you need to delete something unnecessary from it

ldwnt, 2021-09-10:
> The root partition has no free space because it is squashfs and read-only.
> The /tmp partition is insufficient for zip itself and for unpacking it.
> You can download the archive to /data, and unpack to /tmp or to /data .
> But it's better to use a script that will do everything yourself

I execute the commands below using bin files uploaded to /data/tmp.  A write error occurs when the bluetooth firmware is updated. Shall I restart the device or update the bluetooth firmware again?
```
# run_ble_dfu.sh /dev/ttyS1 /data/tmp/full_ble_1.5.0_0102.gbl 150 1
uart-dfu /dev/ttyS1 /data/tmp/full_ble_1.5.0_0102.gbl 1 OTA_STATE:0
do app!!
start update ble firmware!!
killall: start_ot.sh: no process killed
ota_state.txt is exist
killall: ncp_daemon: no process killed
sh: write error: Device or resource busy
in bootloader
Bytes to send:377280
Syncing..ID:000100a0
..DFU OK
Bootloader version: 17301508 (0x1080004)
DFU packet size:377280
99%
finish
ncp target version:150
--200--->app ota_state:0
----->ble app updated

# fw_update /data/tmp/linux_1.5.0_0102.bin
Info: Open /data/tmp/linux_1.5.0_0102.bin, size: 2157588
Info: Firmware type: RTX Linux
Info: data_sum: cad3
Info: Upgrade linux bank 1
Info: Write to /dev/mtdblock7, size:2157572 ...
Process: 100%
Done
Info: Verify ...
Process: 100%
Success

# fw_update /data/tmp/root_1.5.0_0102_mod20210816.bin
Info: Open /data/tmp/root_1.5.0_0102_mod20210816.bin, size: 13529106
Info: Firmware type: RTX Root
Info: data_sum: 6c5a
Info: Upgrade rootfs bank 1
Info: Write to /dev/mtdblock8, size:13529090 ...
Process: 100%
Done
Info: Verify ...
Process: 100%
Success

```

zvldz, 2021-09-10:
```
sh: write error: Device or resource busy
```
It's normal, don't mind it.
The firmware update was successful.

ldwnt, 2021-09-10:
> ```
> sh: write error: Device or resource busy
> ```
> 
> It's normal, don't mind it.
> The firmware update was successful.

It works! Thank you very much~

closed 2021-09-10 by ldwnt