zvldz/mgl03_fw #12: The gateway is not added to the Mi Home if the region China is selected

sanfweb, 2021-11-09:
I installed the stock firmware (firmware/stock/mgl03_1.5.0_0102/) on my gateway, now I can add a gateway to Mi Home only if the region of Russia is selected, if I choose the region of China, then the process of adding to Mi Home is not completed, how can I fix it? all other devices I have in the China region.

zvldz, 2021-11-09:
Try doing a factory reset.
10 clicks on the button.
Maybe your gw is still attached to region RU.

sanfweb, 2021-11-11:
thanks for the answer)

I did a full reset (10 clicks on the button) several times, after that I tried again to add a gateway through the Mihome application with the China region, it freezes at step 2 on timeout. If I reset the gateway again (10 clicks on the button), I change the region to RU and it is perfectly added via Mihome.

zvldz, 2021-11-11:
Very strange. 
Go to our Telegram group https://t.me/xiaomi_gw_v3_hack , maybe we can help you there.

sanfweb, 2021-11-11:
ok, thanks, I wrote there

closed 2022-02-22 by zvldz