zvldz/mgl03_fw #14: 1.5.4_0052

neoghy, 2022-02-10:
the new version is 1.5.4_0052   can i  update ?,,thx

zvldz, 2022-02-10:
I have still not been unable to get this firmware ((

zvldz, 2022-02-22:
Custom and stock firmware 1.5.4_0052 is available.
If you're using https://github.com/AlexxIT/XiaomiGateway3, it's best not to rush it just yet.

Psytoshgen, 2022-03-05:
just updated, 11 days after "not to rush" message... and i cant connect XG3 to ZHA... any ideas? HA is my Free Time toy, so its not critical.
Алекс, доброго времени суток, понимаю, что сейчас вам точно не до всей этой возни. Использую HA как очень интересное занятие для себя в свободное время. Ждал обновления кастомной прошивки, прошился, без ошибок и проблем на последнюю. Но не могу понять почему не цепляется к ZHA, думал про Z2MQTT но пока не мой уровень, ко всему надо по порядку подходить. Что пробовал. сразу ставить данные  во всплывающем окне Socket://итд пробовал автоматическое определение... результат один и тот же.  connection refused. Использую облако nabu casa, вообще заметил, что куча проблем именно в облаке. Но ZHA вроде как локальная интеграция. Допускаю проблему в своих конфигах... но не знаю куда копать. Заранее спасибо.

neoghy, 2022-03-24:
ok i know thx

closed 2022-04-10 by zvldz