zvldz/mgl03_fw #18: Updating zigbee firmware of mgl03 gateway via telnet for Zigbee2MQTT

acrueda, 2022-07-26:
Hello, good morning, I can't update Gateway 3 with firmware for Zigbee2MQTT, it doesn't give any error, but it doesn't continue, I attach a screenshot.
Is it better to use this firmware to use Zigbee2MQTT right now I have mgl03_1.5.0_0102_mod20210816 ?
Thank you so much
[image]

trytocry, 2023-01-27:
same for me  stuck in bootloader

Xmister, 2023-03-21:
I was facing a similar issue, and found out a few things:
* XiaomiGateway3 addon has to be running in HA, otherwise some needed processes won't start
* In the default Zigbee mode however `Lumi_Z3GatewayHost_MQTT` will interfere with the flash process so it will never finish.
* If you set up Zigbee2Mqtt in HA (and keep the XiaomiGateway3 addon running), it will be in the right state for the flash to finish.

gnalbandian, 2023-05-05:
Same here as @kkmaslowski. I think I broke the bootloader. I got a bootloader message previously when minicom into dev/ttyS2 but nothing now. 
Is there anyway in which to flash the bootloader again to the module?

kkmaslowski, 2023-05-05:
@gnalbandian AlexxIT fixed it remotely on my gateway. We tested that flashing zigbee firmware from HA GUI works normally on HassWP (https://github.com/AlexxIT/HassWP - HA with AlexxIT addon running on windows) and don't work on my HA in docker. He also fixed flashing zigbee firmware when current version is unknown (like mine above) in commit: https://github.com/AlexxIT/XiaomiGateway3/commit/ebae56f907b6e446408d4739d94e825f6c25d496

I think you should try to do the following steps:
- download HassWP on Windows: https://github.com/AlexxIT/HassWP
- replace custom_components/xiaomi_gateway3/core/ezsp.py with fixed version https://github.com/AlexxIT/XiaomiGateway3/blob/ebae56f907b6e446408d4739d94e825f6c25d496/custom_components/xiaomi_gateway3/core/ezsp.py
- add GW to it 
- try to flash zigbee firmware from HA GUI

What is your setup? You also run HA in Docker?