zvldz/mgl03_fw #21: Unable to flash custom firmware for Xiaomi Multimode Gateway 2 CN (DMWG03LM)

terenceng81, 2023-01-24:
Not sure why, this does not happen on my Xiaomi Multimode Gateway 1 CN (ZNDMWG03LM)

[image]

/home # wget -O /tmp/curl "http://master.dl.sourceforge.net/project/mgl03/bin/cu
rl?viasf=1" && chmod +x /tmp/curl
Connecting to master.dl.sourceforge.net (216.105.38.12:80)
curl                 100% |*******************************|  2271k  0:00:00 ETA
/home # export PATH="$PATH:/tmp"
/home # curl -s -k -L -o /tmp/update.sh https://raw.githubusercontent.com/zvldz/
mgl03_fw/main/firmware/mgl03_update.sh && sh /tmp/update.sh
/tmp/curl: line 1: syntax error: unexpected "("
/home #

Andrei-Zayats, 2023-07-02:
I have the same issue on my Xiaomi Smart Home Hub 2 (ZNDMWG04LM)
Firmware 1.0.5_0016

[image]

```
/ # wget -O /tmp/curl "http://master.dl.sourceforge.net/project/mgl03/bin/curl?viasf=1" && chmod +x /tmp/curl
Connecting to master.dl.sourceforge.net (216.105.38.12:80)
curl                 100% |***************************************************************************************************|  2271k  0:00:00 ETA
/ # export PATH="$PATH:/tmp"
/ # curl -s -k -L -o /tmp/update.sh https://raw.githubusercontent.com/zvldz/mgl03_fw/main/firmware/mgl03_update.sh && sh /tmp/update.sh
/tmp/curl: line 1: syntax error: unexpected "("
/ # 
 
```

zvldz, 2023-07-02:
This firmware is only for Xiaomi Gateway 3(lumi.gateway.mgl03), not for Xiaomi Smart Multi-Mode Gateway 2(lumi.gateway.mcn001).

cas-class, 2024-11-16:
Greetings, on my ZNDMWG04LM, I get the same error. Is it possible to do something?

Andrei-Zayats, 2024-11-20:
> Greetings, on my ZNDMWG04LM, I get the same error. Is it possible to do something?

I bought Xiaomi Gateway 3 :)