zvldz/mgl03_fw #22: Xiaomi gateway 3 ethernet

DrArut, 2023-01-26:
Hello. Thank you for your work. 
I soldered the rj45 connector to Xiaomi Gateway 3. But I can’t figure out how to make the gateway automatically connect to ethernet instead of wifi. There is only one way on the ethernet using telnet, but it does not suit me, because after each restart I need to enter this command. How can I change the firmware of the gateway so that it automatically turns on ethernet? 
Thanks

Nicols07, 2023-04-23:
[image]

szymucha94, 2023-12-26:
> [image]

Don't do just that :) One of the daemons will sooner or later detect that wlan0 is down and will reenable it... causing both wlan0 and eth0 to be active. If set to the same MAC and/or IP to both (and you will, because both are under br0) it will loop packets and paralyze your network, or at least the gateway.
My take on that was to modify the daemon to never touch wlan0, also to get rid of all unnecessary xiaomi processes as I was using z2m anyway.

Tom-Jojo, 2024-06-23:
> Don't do just that :) One of the daemons will sooner or later detect that wlan0 is down and will reenable it... causing both wlan0 and eth0 to be active. If set to the same MAC and/or IP to both [...]

Hello, could you share your settings for stable Ethernet connevtivity?

Tom-Jojo, 2024-06-23:
> Don't do just that :) One of the daemons will sooner or later detect that wlan0 is down and will reenable it... causing both wlan0 and eth0 to be active. If set to the same MAC and/or IP to both [...]

Hello @szymucha94, could you share your settings for stable Ethernet connectivity?

szymucha94, 2024-06-23:
> Hello @szymucha94, could you share your settings for stable Ethernet connectivity?

by settings you mean my script? This is the run.sh script I have in my notes.
```
/bin/dropbear -R -B
sleep 10
ifconfig eth0 hw ether <mac>
ifconfig eth0 up | ifconfig wlan0 down
sleep 5
if ping -c 1 <ip_of_router> -W 2 > /dev/null; then
    echo "eth0 is up and working, leaving wlan0 disabled" > /tmp/eth0enable.log
    touch /tmp/dont_need_monitor_wpa_supplicant
    /data/wifi_start.sh &
else
    echo "eth0 is not working, re-enabling wlan0" >> /tmp/eth0enable.log
    ifconfig wlan0 up | ifconfig eth0 down
fi
```
Do not use script above if you don't know what you're doing.

You also need to modify the daemon bash script that checks wlan0 status every 15(?) minutes, copy it into persistent storage and delete the whole if statement. Then if you run it it should prevent startup of original script from ROM.
 
Also ethernet is unstable on these units and should not be used for extended periods of time. And it's not just bash daemon that restarts wlan, it's also miio_agent itself. So unless you're ready to de-cloud the gateway and use z2m only (without alexxit integration) just stick to wifi.

Johnnybyzhang, 2026-08-29:
I maintain a fork that now solves this at the firmware level:

https://github.com/Johnnybyzhang/mgl03_fw

It builds on this repository and the earlier MGL03 hardware research. Normal gateway traffic uses eth0. It does not depend on a /data/run.sh interface toggle or a Home Assistant watchdog.

wlan0 remains available for one purpose. After a factory reset, the gateway starts the stock _mibt onboarding AP so Mi Home can configure and bind it again. Once pairing finishes and wired DHCP is available, that AP stops.

I tested 1.5.7_0001 on two separate MGL03 gateways. Both completed a factory reset and Mi Home re-pairing. I also use the result with AlexxIT/XiaomiGateway3.

The rootfs patches the transport-facing interface strings in basic_gw, homekitserver, and zigbee_gw where present. On firmware 1.5.4 and later, it also fixes a four-byte pairing bug in silabs_ncp_bt that could generate malformed JSON when ln_type was absent.

Hardware warning: use a 10/100 magjack with integrated Ethernet magnetics. A passive RJ45 caused connection problems in my test. These builds have no Wi-Fi fallback.

Only 1.5.7_0001 has completed full hardware testing so far. The older firmware bases build and pass the offline suite, but remain experimental.

Build scripts, firmware, hashes, documentation, and full credits are in the repository.