zvldz/mgl03_fw #25: Can't flash Zigbee Firmware

Andreiva13, 2023-06-30:
Hi!
Trying to update the Zigbee Fimware fails both from the Xiaomi3 HA Integration and telnet and I think I bricked it since zigbee2mqtt can't connect anymore.
Output from telnet firmware update attempt:
`# sh /tmp/mgl3_zigbee_flash.sh

Xiaomi Gateway 3 (mgl3) Zigbee firmware flashing script

Found firmware files:
1: ncp-uart-sw_mgl03_6_6_2_stock.gbl
2: ncp-uart-sw_mgl03_6_7_10_z2m.gbl
3: ncp-uart-sw_mgl03_6_7_8_z2m.gbl
Select firmware: 2
You selected: ncp-uart-sw_mgl03_6_7_10_z2m.gbl

Preparing
Killing default zigbee software
killall: socat: no process killed
killall: daemon_app.sh: no process killed
Done

Press Ctrl+C to cancel in 5 seconds!

Detected EZSP v7
Rebooting to bootloader
Sending upload command
Sending ncp-uart-sw_mgl03_6_7_10_z2m.gbl, 1548 blocks: Give your local XMODEM receive command now.
Xmodem sectors/kbytes sent:   0/ 0kRetry 0: Got 65 for sector ACK
Retry 0: NAK on sector
Xmodem sectors/kbytes sent: 122/15kRetry 0: Got 0d for sector ACK`

Ostromogilski, 2023-11-15:
I'm facing the same issue. Don't know what causing this

vrtlspd, 2024-03-23:
I'm in the same boat as you. What are we doing wrong?

```
# wget -O /tmp/zigbee_flash_mgl03.zip "http://master.dl.sourceforge.net/project/mgl03/zigbee/zigbee_flash.zip?viasf=1" && unzip -o /tmp/zigbee_fla
sh_mgl03.zip -d /tmp && cd /tmp && sh /tmp/mgl3_zigbee_flash.sh
Connecting to master.dl.sourceforge.net (216.105.38.12:80)
zigbee_flash_mgl03.z 100% |*************************************************************************************************|   426k  0:00:00 ETA
Archive:  /tmp/zigbee_flash_mgl03.zip
  inflating: mgl3_zigbee_flash.sh
  inflating: ncp-uart-sw_mgl03_6_6_2_stock.gbl
  inflating: ncp-uart-sw_mgl03_6_7_8_z2m.gbl
  inflating: ncp-uart-sw_mgl03_6_7_10_z2m.gbl

Xiaomi Gateway 3 (mgl3) Zigbee firmware flashing script

Found firmware files:
1: ncp-uart-sw_mgl03_6_6_2_stock.gbl
2: ncp-uart-sw_mgl03_6_7_10_z2m.gbl
3: ncp-uart-sw_mgl03_6_7_8_z2m.gbl
Select firmware: 2
You selected: ncp-uart-sw_mgl03_6_7_10_z2m.gbl

Preparing
Killing default zigbee software
killall: socat: no process killed
killall: daemon_app.sh: no process killed
Done

Press Ctrl+C to cancel in 5 seconds!

Detected EZSP v7
Rebooting to bootloader
Sending upload command
Sending ncp-uart-sw_mgl03_6_7_10_z2m.gbl, 1548 blocks: Give your local XMODEM receive command now.
Xmodem sectors/kbytes sent:  63/ 7kRetry 0: Got 0d for sector ACK
```

sergiobaiao, 2026-04-02:
did you guys suceed on this?