zvldz/mgl03_fw #30: ncp-uart-sw_mgl03_6_6_2_stock.gbl for return to stock firmware ???

akimovaleks, 2024-10-04:
` wget -O /tmp/zigbee_flash_mgl03.zip "http://master.dl.sourceforge.net/project/
mgl03/zigbee/zigbee_flash.zip?viasf=1" && unzip -o /tmp/zigbee_flash_mgl03.zip -
d /tmp && cd /tmp && sh /tmp/mgl3_zigbee_flash.sh
Connecting to master.dl.sourceforge.net (216.105.38.12:80)
zigbee_flash_mgl03.z 100% |*******************************|   426k  0:00:00 ETA
Archive:  /tmp/zigbee_flash_mgl03.zip
  inflating: mgl3_zigbee_flash.sh
  inflating: ncp-uart-sw_mgl03_6_6_2_stock.gbl
  inflating: ncp-uart-sw_mgl03_6_7_8_z2m.gbl
  inflating: ncp-uart-sw_mgl03_6_7_10_z2m.gbl
Downloading sx...curl doesn't exists, downloading...done!
chmod: /data/sx: No such file or directory
done!
Downloading libnsl.so.0...chmod: /data/libnsl.so.0: No such file or directory
done!
chmod: /data/sx: No such file or directory

Xiaomi Gateway 3 (mgl3) Zigbee firmware flashing script

Found firmware files:
/data/busybox: line 1: Redirecting: not found
/data/busybox: line 1: Redirecting: not found
/data/busybox: line 1: Redirecting: not found
Select firmware:
`

TAQXYZ, 2024-10-14:
me too，Please help me

fifammlaf, 2024-12-30:
try running
`curl -k -o /data/busybox https://busybox.net/downloads/binaries/1.16.1/busybox-mipsel`

as the /data/busybox file is somehow `Redirecting`